madl3x/x-scara #1: Typos in te arm BOM and numer of print pieces

sl1pkn07, 2020-11-07:
Hi

i've found some typos in the BOM and in the numer of print pieces in the ARM

[image]

but in the instructions show 2

[image]
[image]

in the BOM sections, show need a 20t(eet) GT2 for 5mm bore

[image]

but in the instruction, show a 20t FT5 8mm bore
[image]

this is not expecified in the BOM, for abois missustanding (some kits sell 60t pulleys (8mm bore) + 280mm belts + 20t pulley but with 5mm bore instead of 8mm. i this this need specified in the bom

greetings

sl1pkn07, 2020-11-07:
Hi

also found a missing belt. is the same size like the two others?

greetings

amircescu-gpfw, 2020-11-08:
> Hi
> 
> also found a missing belt. is the same size like the two others?
> 
> greetings

Yes, they are 3 belts in total, all the same length. 
Thanks for noticing the BOM problems, I will be fixing them right away.

madl3x, 2020-11-08:
Thanks for noticing the issue. I will verify the BOM and fix it right away. 

> also found a missing belt. is the same size like the two others?
Yes, they are 3 and all are 280mm belts.

sl1pkn07, 2020-11-08:
Hi again. durning print, ive found sommeting issues.

the `Arm_Shoulder_Connector`, the screws to fix the bearing arm, seems so be a little more lengeith, 12mm instead of 10mm, because with 10mm the screw is hidde in te piece instead rebase to grip the bearing

[image]

and in the `Arm_Base_ZHub_Top` and `Arm_Base_ZHub_Bottom`, the holes for the screws not grip the screws. i'm not sure is is a tolerance problem or by design. ive fixing it driling deep the holes and use large screws

greetings

sl1pkn07, 2020-11-08:
[image]
 K -> C

XD

madl3x, 2020-11-13:
> the `Arm_Shoulder_Connector`, the screws to fix the bearing arm, seems so be a little more lengeith, 12mm instead of 10mm, because with 10mm the screw is hidde in te piece instead rebase to grip the bearing

I checked the design and indeed those are M3 x 12mm not M3 x 10 mm. The documentation for this part wrongly says all screws are M3 x 10 for this part. Front holding the front bearing please use 12 mm screws.

> and in the Arm_Base_ZHub_Top and Arm_Base_ZHub_Bottom, the holes for the screws not grip the screws. i'm not sure is is a tolerance problem or by design. ive fixing it driling deep the holes and use large screws

Are you talking about Z-nut screws that enter the `Arm_Base_Zhub`? Those have a 3.1 mm diameter in design and should hold M3 screws pretty well. If you are slicing with Cura, please check the `Horizontal Expansion` setting. It may help you get the part printed right.

madl3x, 2020-11-13:
> [image]
> K -> C
> 
> XD

Yes, correct, please create a different issue for that, so we can easily track each problem.

sl1pkn07, 2020-11-13:
> > the `Arm_Shoulder_Connector`, the screws to fix the bearing arm, seems so be a little more lengeith, 12mm instead of 10mm, because with 10mm the screw is hidde in te piece instead rebase to grip the bearing
> 
> I checked the design and indeed those are M3 x 12mm not M3 x 10 mm. The documentation for this part wrongly says all screws are M3 x 10. For holding the front bearing please use M3 x 12 mm screws.

yes. i see that and i fixed in my build

some other pieces need M3x8 instead M3x10

-  for join `Arm_EMount_Fan` to `Arm_EMount_Base_Left `+ `Arm_EMount_Base_ABL`
-  for hold Z Nema17 to pieces
-  for hold the 8x60mm calibrated row to `Arm_Elbow_RodHub` (the screw not thread all, but better than M3x10

and M3x15

-  for hold the  `Arm_Elbow_RodHub` to `Arm_Elbow_Plate`s

the `Arm_Base_MotorPlate` is now in print process

> Are you talking about Z-nut screws that enter the `Arm_Base_Zhub`? Those have a 3.1 mm diameter in design and should hold M3 screws pretty well. If you are slicing with Cura, please check the [...]

the holes should be small than the outter periferic thread of the screw. the drill (normalized, i think) for making M3 threads is 2.5 or 2.25mm/diameter (my thread driver kit have this drill) (https://fullerfasteners.com/tech/recommended-tapping-drill-size/ for example). yes, is possible fix it on the fly with the vertical_expansion (i use superslicer, (from that piece i set 0.1 that value), but i think should be fixed in the STL

> Yes, correct, please create a different issue for that, so we can easily track each problem.

ok

greetings

sl1pkn07, 2020-11-13:
M3x8
[image]
[image]

M3x8 (center) and M3x15
[image]

sl1pkn07, 2020-11-14:
First `Arm_Base_MotorPlate` printed

-  the screws for grab the nema17 should be M3x8mm (and washer) instead of M3x10
-  if the screws for grab the GT2 idlers is M3, the GT2 idlers should be with bore 3mm instead 5mm. or need make a caps?

greetings

madl3x, 2020-11-21:
> * the screws for grab the nema17 should be M3x8mm (and washer) instead of M3x10
I'll keep that in mind but mostly depends on the motor you have. Some NEMA17 motors have deeper threads for screws. The `Arm_Base_MotorPlate` part is 4 mm thick where it connects with the motor and therefore 6mm from the screw's thread will go into the motor itself. But some motors may not accommodate that. 

> if the screws for grab the GT2 idlers is M3, the GT2 idlers should be with bore 3mm instead 5mm. or need make a caps?
That is another good remark. The GT2 idlers I used are 3mm ID, not 5mm inner bore like the BOM says. I will correct that and close this issue when I fix the BOM.