madl3x/x-scara #2: E3D hotend support not fit in the elbow top/bottom plates

sl1pkn07, 2020-11-13:
[image]
[image]
[image]

madl3x, 2020-11-13:
Thank you, your reported issue is correct. There is an overlap of 1.61 mm between the parts.
I will fix it and update the STL and .f3d files. Until then please just trim your part or cut the STL with an editor, so you can finish the assembly.

sl1pkn07, 2020-11-13:
thanks, but i don't have hurry at all, the frame prefix and the most mechanic parts is still on transit (ETA end of november or start of december)

so take you time

greetings

closed 2020-11-28 by madl3x