madl3x/x-scara #3: Typo in the Frame_Plate_SideConnector piece

sl1pkn07, 2020-11-13:
[image]
 K -> C

greetings

sl1pkn07, 2020-11-21:
Hi

for print this piece, in the right side, is needed to print it in mirror mode, but the text is also morrored

[image]
[image]

for look&feel better, i think this piede needs to be added in the piece list a mirrored piece but with thext in the rigth position

what you think?

greetings

madl3x, 2020-11-28:
Fixed typo. Added two separate versions (left and right) for the side connectors. Thank you for noticing.

closed 2020-11-28 by madl3x