madl3x/x-scara #4: [INFO] Electronics, Endstops and Marlin, How do it

sl1pkn07, 2020-11-19:
Hi, my setup is almst done (missing frame pieces), but i can adapt the lectronics to the machine

what is the minimun requirements (inputs/outputs) for this project?, (a Fysetc AIOII https://github.com/FYSETC/FYSETC-AIO_II is sufficient? (with TMC2208, Silent, but no sensorless :( )

where i can set the axis endstops in the machine? can you share you endstops mounts? (i see something in the instagram vids, but i'm not clear)

in marlin (2.0.x) exist config for SCARA machines, one is `MP_SCARA`, and other is `Morgan_SCARA`, what is the best fit for this machine?

greetingsd

madl3x, 2020-11-21:
Hello,

Endstops is the last missing piece of this machine and that was what I was working on. Current design is missing that. Technically speaking, you can use normal mechanical endstops and mount them on Arm Base part like all the other SCARA models do. That's how both Morgan and MP Scara works as well.

My design (what you've seen in the Instagram videos) uses something else. It's a center magnetical endstop-like device which Marlin does not support yet. The "endstop" serves as a sensor which tells the machine where the arm shoulder and arm elbow are fully extended (i.e. the angle they make relative to machine's front axis  is 0º). This is not yet implemented in Marlin and I've worked on that but not fully finished. I need to rewrite the home sequence (G28) to make that work.

Untill then, you can use mechanical endstops connected to the Arm Base part. 

Regarding which model to use (Morgan or MP_SCARA), I started from Morgan Scara but rewrited the inverse kinematics code for that to work. I have that completed already and I will post an update with an Alpha firmware to have access to the code, soon. This will make your machine move in the X,Y cartesian space correctly.

sl1pkn07, 2020-11-29:
Hi. my printer is almost done

i have used the marlin Morgan configuration adapted to my board, and need apply (bugfix-2.0.x) this patchs

https://github.com/annhan/Configurations/commit/1c23a536228de459c9f6fc3cb4811e3a498e3bba
https://github.com/MarlinFirmware/Marlin/pull/17274

but not work at all (only home X and Y, the rest fails)

i use Hall effect Endstops (KY-003) fixed (with little screws) the Arm_SideWall pieces in the Elbow and Shoulder, activated with little neodimium magnets (one fixed in the Arm_Base_MotorPlate, and other in the bottom Arm_Shoulder_Connector). this works as spected

question. what mean "THETA" and "PSI" in this environment?

greetings

sl1pkn07, 2020-11-29:
[VID_20201129_224254.zip](https://github.com/madl3x/x-scara/files/5612902/VID_20201129_224254.zip) (rename to 3gp)
[image]
[image]

madl3x, 2021-01-15:
@sl1pkn07 You might wanna take a look at the first firmware release I made. 
https://github.com/madl3x/x-scara/tree/master/firmware

Version 0.1 works without X,Y endstops (for the moment), with manual calibration (use M362, check the Gcodes guide I've published). I am going to release 0.2 soon (fixes the orientation of the cartesian space) and after that, I will implement homing, end-stops mounting and bed leveling.

I just ordered the same endstops you have, and got the package today, so I will be able to work on that soon. I like your positioning but I have another idea in mind. Keep you posted.

sl1pkn07, 2021-01-15:
the positionn is reversed. my phone break that day and i take the photos with the tablet frontal cam

amircescu-gpfw, 2021-01-15:
Off-topic: @sl1pkn07 please post a make on thingiverse, if your machine is ready. I would love to see more pictures of it and in more detail.
On-topic: I will get back to endstops after i make this second update on the firmware.

sl1pkn07, 2021-01-17:
Hi

i've apply you changes for Xscara (based in marlin 2.5.0.2(?)) to upstream marlin bugfix-2.0.x branch

steps i do:

- download you repo
- download marlin by git point to 2.0.5.2 tag
- copy you marlin to cloned upstream marlin
- review the changes
- git diff > patch.patch
- revert the changes and checkout to bugfix-2.0.x branch
- apply the patch created before
- fix the rejected code and review again

~~~patch
diff --git a/.gitignore b/.gitignore
index 62f73a7c0d..88286ebe5d 100755
--- a/.gitignore
+++ b/.gitignore
@@ -23,6 +23,9 @@
 # NEVER put it in the repository
 _Version.h
 
+process-palette.json
+.pio/
+
 #
 # OS
 #
diff --git a/Marlin/src/core/language.h b/Marlin/src/core/language.h
index d6048d293c..af3ee3479d 100644
--- a/Marlin/src/core/language.h
+++ b/Marlin/src/core/language.h
@@ -126,7 +126,11 @@
 #define STR_E_STEPPER_NOT_SPECIFIED         "E stepper not specified"
 #define STR_INVALID_SOLENOID                "Invalid solenoid"
 #define STR_COUNT_X                         " Count X:"
+#if ENABLED(X_SCARA)
+#define STR_COUNT_A                         "\nMotors A:"
+#else
 #define STR_COUNT_A                         " Count A:"
+#endif
 #define STR_WATCHDOG_FIRED                  "Watchdog timeout. Reset required."
 #define STR_ERR_KILLED                      "Printer halted. kill() called!"
 #define STR_ERR_STOPPED                     "Printer stopped due to errors. Fix the error and use M999 to restart. (Temperature is reset. Set it after restarting)"
diff --git a/Marlin/src/gcode/calibrate/M665.cpp b/Marlin/src/gcode/calibrate/M665.cpp
index 557204cc11..a13552fa30 100644
--- a/Marlin/src/gcode/calibrate/M665.cpp
+++ b/Marlin/src/gcode/calibrate/M665.cpp
@@ -62,6 +62,31 @@
 
   #include "../../module/scara.h"
 
+  #if ENABLED(X_SCARA)
+  /**
+   * M665: Set SCARA settings
+   *
+   * Parameters:
+   *
+   * D<bool> - debug on or off (if compiled with X_SCARA_DEBUG)
+   * S<int>  - number of segments per second
+   */
+  void GcodeSuite::M665() {
+
+    #if ENABLED(X_SCARA_DEBUG)
+      if (parser.seenval('D')) {
+        x_scara_debug = parser.value_bool();
+      }
+      SERIAL_ECHOPAIR("Debug:", x_scara_debug, " ");
[... 3304 lines cut ...]
@@ -394,6 +394,7 @@ PRINTCOUNTER            = src_filter=+<src/module/printcounter.cpp>
 HAS_BED_PROBE           = src_filter=+<src/module/probe.cpp> +<src/gcode/probe/G30.cpp> +<src/gcode/probe/M401_M402.cpp> +<src/gcode/probe/M851.cpp>
 IS_SCARA                = src_filter=+<src/module/scara.cpp>
 MORGAN_SCARA            = src_filter=+<src/gcode/scara>
+X_SCARA                 = src_filter=+<src/gcode/scara>
 (ESP3D_)?WIFISUPPORT    = AsyncTCP, ESP Async WebServer
   ESP3DLib=https://github.com/luc-github/ESP3DLib.git
   arduinoWebSockets=https://github.com/Links2004/arduinoWebSockets.git

~~~

some changes is not applied because is not in the code, like

https://github.com/madl3x/x-scara/blob/master/firmware/Marlin-2.0.x/Marlin/src/MarlinCore.cpp#L234 (not in the code in this form)
https://github.com/madl3x/x-scara/blob/e2b34428e86c7210a7e92ed612f71fc7dfde8c0f/firmware/Marlin-2.0.x/Marlin/src/gcode/bedlevel/abl/G29.cpp#L204-L207 (not in the code in this form)

and others i adapted in the code (sanityCheck.h)

i also keep clean `Marlin/Configuration.h` and `Marlin/Configuration_adv.h` if you want merge the code in upstream some day (keeping `config/SCARA/X-SCARA/Configuration.h` but with the essential mechanical structural settings ansd steps/min/deg, leaving user configurations (electronics, accels, jerks, feedrates, hot/bed, etc) untouched.

can you check it? (im not coder, but i know how mod it)

greetings

PD: please, brim back the `platformio.ini`. i can't build the code without it. arduino ide is not a option for me

madl3x, 2021-01-17:
> i've apply you changes for Xscara (based in marlin 2.5.0.2(?)) to upstream marlin bugfix-2.0.x branch
Firmware for X-SCARA is not yet ready for up-stream integration. I will merge it to the upstream when it leaves the alpha stage. For the moment, use it just from this (github) project.

> i also keep clean Marlin/Configuration.h and Marlin/Configuration_adv.h if you want merge the code in upstream some day (keeping config/SCARA/X-SCARA/Configuration.h but with the essential [...]

Thanks, good advice. I will clean it before upstream merge.

> PD: please, brim back the platformio.ini. i can't build the code without it. arduino ide is not a option for me
Sure, I will put it back.

fabtopia, 2021-05-24:
Hi, i have a question, why magnetical endstop en not optical enstops?

sl1pkn07, 2021-05-24:
more or less, the size, and can hide in the structure. but is enterely user preference, optical endtop inplementation is welcome (?)

greetings

PS: i'm not drop the project, but i put it in standby. sorry

madl3x, 2021-05-24:
> Hi, i have a question, why magnetical endstop en not optical enstops?

@fabtopia magnetic endstops don't have the positioning constraints that optical endstops have and they are more precise on repetitive movements. Optical endstops are usually U-shaped and the "flag" that passes through needs to be aligned  with that shape. Magnetical endstops are more tolerant in terms of distance, especially when you use good neodium magnets. 

But, if you have only optical endstops at hand, you can use them instead of the magnetical ones, but you need to come up with your own design for mounting them. For now I just recommend using magnetical endstops for ease of use and better precision over time.

closed 2021-09-05 by madl3x