madl3x/x-scara #5: Missing download button!

alassedy, 2021-01-20:
Hi,
I tried to download the Fusion 360 version of your design, but I couldn't find the download button for the arm assembly when opened this link:
https://a360.co/2YkuguU
Unlike the frame assembly like which has the download button.
Thank you for the great design.

madl3x, 2021-01-31:
I'll verify that, but you can download the STLs from Github or Thingiverse. Check out the releases section.

closed 2021-09-05 by madl3x

jschilli, 2022-12-11:
is it possible to allow downloads of the arm models?  The 'allow item to be downloaded' switch needs to be on in Fusion when sharing the link.

I'm looking to model new end effectors on the arm and it would be helpful to evolve your model vs starting from scratch.

Great project!