madl3x/x-scara #6: Electronics BOM

vojjta, 2021-09-16:
Hi, I really like this project and I want to build it. But in documentation is missing electronics BOM. For example what endstops are used? I know that hall effect sensors but what type, how they are placed... Thank you

madl3x, 2021-09-16:
Hello,

Yes, you are right, I should do an electronics BOM.

Untill then, check out this page:
https://github.com/madl3x/x-scara/tree/master/hardware/arm

This picture to be more specific:
[image: image.png]

I used magnetic endstops and a mount that holds 5x5 mm cylindrical magnets.

This one should work fine:
https://www.my-home-fab.de/en/reprap-3d-hardware/electronics/51/rrd-hall-endstop

You can find other types of magnetic endstops, or you can even make your
own.
If you want to use optical endtsops, you can use them, but you have to
redesign the part that is coupled with the rod, to act as a flag that goes
in the optical sensor.

Regards,
Alex M.

vojjta, 2021-09-18:
Thank you for your answer. 

This one should be also ok?
[Hall magnetic Sensor with analog & digital outs](https://www.aliexpress.com/item/32653988546.html?spm=a2g0o.cart.0.0.3d4a3c00v27LKX&mp=1)

madl3x, 2021-09-18:
It's too long. It will probably fit on top of the arm, but it will not fit on the shoulder joint.
This RADDS Hall-Effect could be a second option:
https://reprapworld.eu/electronics/endstops/hall_e_endstop/#description

You can find suppliers for this one all around the world.

sl1pkn07, 2021-10-18:
if help, my hall units comes from here https://www.ebay.es/itm/221805903273?hash=item33a4a97da9:g:a3cAAOSwnw9Z90Wh (is a local store on my country, no from china (i mean, the delivery)). something similar should be exist in your country (for examplem, from DE https://www.ebay.de/itm/152368482760?hash=item2379de89c8:g:PqQAAOSwOQJeagV8)

greetings

iamthebest22, 2021-11-20:
Sorry for the late question, would this one work? As the RADDS one is out of stock at my location?
https://spool3d.ca/hall-sensor-module-endstop/

amircescu-gpfw, 2021-12-02:
> Sorry for the late question, would this one work? As the RADDS one is out of stock at my location? https://spool3d.ca/hall-sensor-module-endstop/

Yes, it would work. Actually it's the same with the one that I am using.

iamthebest22, 2021-12-02:
Thanks! Now let's hope the newest version of Marlin can do SCARA with his settings fine. I just had to change X-scara to MP_Scara

iamthebest22, 2021-12-05:
> > Sorry for the late question, would this one work? As the RADDS one is out of stock at my location? https://spool3d.ca/hall-sensor-module-endstop/
> 
> Yes, it would work. Actually it's the same with the one that I am using.

nice, how did you mount it? Since it uses two M2 screws, I am not quite sure how to mount it on the elbow one. is there a picture you could show me?

madl3x, 2021-12-05:
Sure check [this pic](https://photos.google.com/share/AF1QipNQx_2L8vYkozSCluqT0DdA4VPUnM8ntkQr1BmR7Fuyo9QzZPlhhxQG7TyLyEseVw?key=YUFLM3diOGdnRkFlRkl2b2MyeER0cDItUUNXdkxR) and [this one](https://photos.google.com/share/AF1QipP8QMgJ6HX2Ko4WXrdBte_jJvhIJaoFje7oFavgWGwPmNMKQPuTCMILyECJUM7MMA?key=VE9YNmQ5VDNGX003aExaWTJVM0pGYlV2RXhzNldB).

I used a strong double adhesive tape along with a tie wrap. 
After the adhesive hardens the tie wrap is pretty much redundant.

iamthebest22, 2021-12-06:
Ohhh that one nice. I should be more clear, I meant the other arm, the one where there's less space. Currently I drilled a small M2 hole and used hot glue, and seems to be well, but would definitely like to see wha tyou use.