madl3x/x-scara #7: BOM

vojjta, 2021-09-19:
Hi,

I went through BOM and there are probably some small mistakes.

In [Arm](https://github.com/madl3x/x-scara/blob/master/hardware/arm/README.md) is Tr8 Z-Nut and in [Frame](https://github.com/madl3x/x-scara/blob/master/hardware/frame/README.md) is T-Nuts. Are those same nuts for lead screw?

In [Frame](https://github.com/madl3x/x-scara/blob/master/hardware/frame/README.md) is 4 pieces for 20x20 60mm aluminum profile. But should not it be 8 pieces(4 for bottom and 4 for top part of frame)?

I am preparing [Sourcing guide](https://docs.google.com/spreadsheets/d/17AkOFRG1sNOwA-htXEAFYQWag8pyb9Ng2faQ-IcJ8a0/edit?usp=sharing) for my self... Would it help also to your project?

madl3x, 2021-09-19:
The Z-nut is [this one](https://www.geeetech.com/images/l/Geeetech_nut.jpg) and the is nut connecting the lead screw to the motor plate.
T-nuts are for frame assembly and they look [like this](https://www.norelem.com/xs_db/BILD_DB/0/www/750/07090.jpg). They are a bit expensive, so for 20x20 aluminum profiles, you can also use square M5 nuts.

```In Frame is 4 pieces for 20x20 60mm aluminum profile. But should not it be 8 pieces(4 for bottom and 4 for top part of frame)?```
That is correct, I will fix that.

```I am preparing Sourcing guide for my self... Would it help also to your project?```
Sure. Let me know when you're done and I will take a look.
It would be good to be in text format, so it can be tracked by git.

madl3x, 2021-09-19:
Fixed with commit 995abc5.

closed 2021-09-19 by madl3x

vojjta, 2021-09-20:
>The Z-nut is this one and is the nut connecting the lead screw to the motor plate.
T-nuts are for frame assembly and they look like this. They are a bit expensive, so for 20x20 aluminum profiles, you can also use square M5 nuts.

As a second point in "Screws and nuts" for frame is 50 x M5 Drop In Hammer T-Nuts for 2020 (OR 50 x M5 square Nuts for 2020) and as a last one is 2 x T-Nuts (corresponding to your chosen lead screw). For Arm in section  "Screws and nuts" is 2 x Tr8 Z-Nut (1 is mandatory, 2 are optional). That is whay I think, that nut for lead screw is mentioned twice