madl3x/x-scara #9: Calibration problem

tsogoo, 2021-10-13:
Hi. I created scara and configured:
```
  // Length of inner and outer support arms. Measure arm lengths precisely.
  #define SCARA_LINKAGE_1  135.0    // (mm)
  #define SCARA_LINKAGE_2  125.0    // (mm)
  
  // SCARA tower offset (position of Tower relative to bed zero position)
  #define SCARA_OFFSET_X     0      // (mm)
```
I approximated :
```
  #define SCARA_OFFSET_Y    -148    // (mm) I approximated it
```
Configured for my gears:
```
  #define DEFAULT_AXIS_STEPS_PER_UNIT   { 46, 40, 400, 418.5 } // For my printers gears
```

Then when I command `M361 S0 E90` Elbow moves 90 degrees
when I command `M361 S45 E0` Shoulder moves 45 degrees but elbow degree around 30 to 40. Is it ok? If not what it should be?

And sent gcode to draw circle but it draws egg shape.

tsogoo, 2021-10-13:
When I command `M361 S45 E0` or `M361 S90 E0` elbow doesn't go 0 degrees. and elbow stepper motor moves.

madl3x, 2021-10-14:
From what I see in the pictures, your model has no mechanical crosstalk between elbow and shoulder, and X-scara does.
X-scara shares with the shounder the elbow reduction gear, which influences the elbow's movement when the shoulder moves.

Try patching `Marlin/src/module/scara.cpp`, and replace lines 339-375 with this:

```C
/* forward kinematics transformation */
static  FORCE_INLINE void x_scara_fwdk(const float &a, const float &b) {
    const float S = a;
    const float E = b-a; // removing crosstalk (/3)

    const float a_sin = - sin(RADIANS(S)) * L1,
                a_cos =   cos(RADIANS(S)) * L1,
                b_sin = - sin(RADIANS(S+E)) * L2,
                b_cos =   cos(RADIANS(S+E)) * L2;

    cartes.set(a_sin + b_sin + scara_offset.x,
               a_cos + b_cos + scara_offset.y); 

    X_SCARA_DEBUG_LNPAIR(
      "FWDK S:", S," E:", E," -> X:", cartes.x," Y:", cartes.y);
}

/*  inverse kinematics transformation */
static FORCE_INLINE void x_scara_invk(const xyz_pos_t &raw) {

    const float x = - (raw.x - scara_offset.x);
    const float y =   (raw.y - scara_offset.y);

    float x2y2 = sq(x) + sq(y);
    float hypot = SQRT(x2y2);

    if(hypot == 0) hypot = 0.000001; // avoid division by zero

    float S = ATAN2(x,y) - ACOS((x2y2 + PK_DifLenSqrd )/(2*L1*hypot));
    float E =              ACOS((x2y2 - PK_SumLenSq   )/(PK_ProdOfLen)) + S; // removing crosstalk (/3)

    delta.set(DEGREES(S), DEGREES(E), raw.z);

    X_SCARA_DEBUG_LNPAIR(
      "INVK X:",raw.x," Y:", raw.y, " -> S:", DEGREES(S)," E:", DEGREES(E));
}
```

tsogoo, 2021-10-25:
So I can choose `SCARA_OFFSET_Y` in specific range. It doesn't effect to drawing shape (circle to ellipse). right?
Maybe perfect elbow angle (SCARA_OFFSET_Y value) to increase quality, right?

I have also one problem. I fixed drawing egg problem. But now I have an ellipse issue. I think I configured arm lengths correct.
And my models shoulder, elbow arm goes right angles (Goes 90 degrees approximately, I used formula but didn't go 90 degree correctly. So I had to use command `M92 X Y` to calibrate angles.).

ToonKru3, 2022-03-09:
i have i same problem have here when i calibrate  move stright manually and then `M362 E0 S0` and `G0 X0 Y0` and i try test move x and y button by Repetier-Host is curve shape to trapezoid

ToonKru3, 2022-03-09:
Can you re-write step by step for calibrate abotu how to fix trapazoid. I have followed by your instruction but it's trapzoid (square shape) and oval (circle shape)

ToonKru3, 2022-03-09:
or some much detail video about how to set up step by step