madl3x/x-scara #12: Board not in the beta firmware, so having issues

iamthebest22, 2021-11-26:
I'll try my best to explain this, the beta firmware you've provided, unless I've missed something, doesn't have the BigTreeTech 32 bit SKR V2.0 board, so I had to download the newest 2.0.9.2 marlin firmware, and then made changes. some of the errors that it gave me before I changed it to "fix" them were like

"SCARA is now Morgan_scara"

so what do now do now? This works, but I'm pretty sure is not what you've meant. Thanks

iamthebest22, 2021-11-26:
Here is the whole configuration.h
/**
 * Marlin 3D Printer Firmware
 * Copyright (c) 2020 MarlinFirmware [https://github.com/MarlinFirmware/Marlin]
 *
 * Based on Sprinter and grbl.
 * Copyright (c) 2011 Camiel Gubbels / Erik van der Zalm
 *
 * This program is free software: you can redistribute it and/or modify
 * it under the terms of the GNU General Public License as published by
 * the Free Software Foundation, either version 3 of the License, or
 * (at your option) any later version.
 *
 * This program is distributed in the hope that it will be useful,
 * but WITHOUT ANY WARRANTY; without even the implied warranty of
 * MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
 * GNU General Public License for more details.
 *
 * You should have received a copy of the GNU General Public License
 * along with this program.  If not, see <https://www.gnu.org/licenses/>.
 *
 */
#pragma once

/**
 * Configuration.h
 *
 * Basic settings such as:
 *
 * - Type of electronics
 * - Type of temperature sensor
 * - Printer geometry
 * - Endstop configuration
 * - LCD controller
 * - Extra features
 *
 * Advanced settings can be found in Configuration_adv.h
 */
#define CONFIGURATION_H_VERSION 02000902

//===========================================================================
//============================= Getting Started =============================
//===========================================================================

/**
 * Here are some useful links to help get your machine configured and calibrated:
 *
 * Example Configs:     https://github.com/MarlinFirmware/Configurations/branches/all
 *
 * Průša Calculator:    https://blog.prusaprinters.org/calculator_3416/
 *
 * Calibration Guides:  https://reprap.org/wiki/Calibration
 *                      https://reprap.org/wiki/Triffid_Hunter%27s_Calibration_Guide
 *                      https://sites.google.com/site/repraplogphase/calibration-of-your-reprap
 *                      https://youtu.be/wAL9d7FgInk
 *
 * Calibration Objects: https://www.thingiverse.com/thing:5573
 *                      https://www.thingiverse.com/thing:1278865
 */

//===========================================================================
//============================= SCARA Printer ===============================
//===========================================================================

/**
 * X-SCARA is developed by Alex Mircescu in 2021 and is an open source 
 * SCARA machine with an unique mechanical design, intented to simplify the
 * calibration process and to improve overall mechanical performance and 
 * robustness for a serial SCARA machine.
 * 
[... 2960 lines cut ...]
  #define PRINTER_EVENT_LEDS
#endif

/**
 * Number of servos
 *
 * For some servo-related options NUM_SERVOS will be set automatically.
 * Set this manually if there are extra servos needing manual control.
 * Set to 0 to turn off servo support.
 */
//#define NUM_SERVOS 3 // Note: Servo index starts with 0 for M280-M282 commands

// (ms) Delay before the next move will start, to give the servo time to reach its target angle.
// 300ms is a good value but you can try less delay.
// If the servo can't reach the requested position, increase it.
#define SERVO_DELAY { 300 }

// Only power servos during movement, otherwise leave off to prevent jitter
//#define DEACTIVATE_SERVOS_AFTER_MOVE

// Edit servo angles with M281 and save to EEPROM with M500
//#define EDITABLE_SERVO_ANGLES

// Disable servo with M282 to reduce power consumption, noise, and heat when not in use
//#define SERVO_DETACH_GCODE

madl3x, 2021-12-04:
Yes, X-SCARA Firmware is derived from 2.0.5.1 that doesn't have support for many SKR boards.
I am working on merging what I did with Marlin 2.0.9.2. There are a lot of changes between these two versions.

iamthebest22, 2021-12-04:
> Yes, X-SCARA Firmware is derived from 2.0.5.1 that doesn't have support for many SKR boards. I am working on merging what I did with Marlin 2.0.9.2. There are a lot of changes between these two versions.

That's good to hear. I have the SKR 2 board, and that actually got a chip change due to chip shortages. Marlin just got it working so that it can compile for the new chip on it. I confirmed it compiles too. Pins are the same. So there's that too. But I think the only thing that matter is just the configuration.h and the configuration_adv.h so I could just copy that into the latest nightly build that supports the SKR 2. 

Base off your reply, does that mean the current nightly marlin build that supports SKR 2 with the new chip not work?
I change Define_X-Scara to Define_MP Scara. 

Thanks.

iamthebest22, 2021-12-06:
just a quick update, some weird issue with the combined nightly build I have that works with my SKR 2 with new chip. Z movement is fine when I just want to move it up and down in the correct direction. However, when I go to home, it moves in the wrong direction and then just kills says printer halted. Endstops are set up correctly.

madl3x, 2021-12-16:
Are you using X-SCARA Marlin 2.5.1 or MP Scara?

iamthebest22, 2021-12-16:
huh? I'm using the newest bugfix nightly build version, it's not that, Marlin doesn't play well with Precision Piezo Orion. I'm switching to Reprapfirmware now as that works with the probling system that i use. If I have your permission, once I get it sorted on RRF, I'll release the configs needed so there's options for marlin or RRF.

amircescu-gpfw, 2021-12-17:
Sure, that would be very interesting. Never tried ReprapFirmware for this model.
Let me know how it works.

iamthebest22, 2021-12-21:
Got good news and once "big" bad news.
RRF is working, still sorting out some setup, but it works in general. However it is Xmas so things are slow. Everything homes and movement all works.

The bad news? I tried using precision piezo orion, which has the nozzle as the probe. Unfortunately maybe it's me etc, but the arm is just too lax that there's too big of a deviation when I do probe repeatability. Have to use a BLtouch or something similar. 

Now if you look down here at this site: https://miscsolutions.wordpress.com/2017/09/24/improving-the-robotdigg-scara-printer/ He did this actually a few years ago, but he does says this if you read the leveling section

It must be possible to mount it very close to the nozzle. In most printers, XY offset between the Z probe and the nozzle can be compensated. But in a SCARA printer, the print head rotates, so the XY offsets are not constant. In principle the Z probe offset from the nozzle could be specified as a modified distal joint angle and a modified distal arm length, so that the probe can be moved to the exact spot required; but RepRapFirmware doesn’t implement this yet.

In your Marlin config, I see you've did it for Marlin in the offset by degrees, which is 20 in this case.

I am asking for Duet to implement this, but I have no idea if we're get this done before work starts again in January. I also don't want to rush you as you are porting your firmware to the newest 2.0.9.2. If I could just make a request to make sure the version you have supports the new BigTreeTech SKR 2 with the 429 chip (NOT the 407 chip). 

Fingers crossed, because everything works, just not the precision piezo orion sensor.

Riverolf, 2022-04-28:
I have been trying to get Marlin 2.0.9 to work, but I can not. Do you have a list of what you had to change to get the 2.0.5.1 to work?

gnmrc-s, 2022-04-29:
I'm trying to use a creality 4.2.7 board with the firmware, it compiles but gives warning for data storage, any idea on how to fix it?
[image]

gnmrc-s, 2022-05-05:
I tried with MP_SCARS configuration but I get a problem while homing, the endstops of link 1 and link 2 work properly during testing with M119, but during homing they do not stop the arms

Riverolf, 2022-05-05:
This is what I had to do:

motion.cpp			//comment out	
#if EITHER(MORGAN_SCARA, MP_SCARA)
      // Only Z homing (with probe) is permitted
      //if (axis != Z_AXIS) { BUZZ(100, 880); return; }
    //#else

configuration.h		//comment out
// Validate that endstops are triggered on homing moves
//#define VALIDATE_HOMING_ENDSTOPS		

configuration.adv		//uncomment
#define QUICK_HOME                     	// If G28 contains XY do a diagonal move first

scara.cpp			//add one line of code
// Angle of Arm2
    PSI = ATAN2(S2, C2);
    PSI = PSI + THETA/3; //RSTROMO add this line

scara.cpp line 75
#if ENABLED(MORGAN_SCARA)

change to: 
#if ENABLED(MP_SCARA)

I also had to disable software endstops to get it printing. Gcode: M211 S0

gnmrc-s, 2022-05-05:
I have done the same config, I think we have followed both the same video, but I do not have undefined the software endstops, maybe this will solve my bug. At the moment I'm reproducing the X-SCARA marlin 2.5.2, on marlin 2.6.0 that is the first firmware that supports my board(creality 4.2.7)

Riverolf, 2022-05-06:
The arm wont take G1 moves when they are outside the print area.
Just add M211 S0 to your startup gcode.

gnmrc-s, 2022-05-06:
maybe I have a problem with the endstops, because my arm won't stop even with your changes.
The thing that I don't understand is that when I check my endstops status with m119, they work properly but when I start the homing procedure the endstops do not activate anymore. It's 2 months that I am stuck in this situation and it's frustrating.

gnmrc-s, 2022-05-07:
I have solved it, well not completely, it's a bug with the firmware that enable the endstops feature only if you home while settings the endstops as MIN endstops.