madl3x/x-scara #13: Top Endstop location wrong?

iamthebest22, 2021-12-08:
Hi Alex, wondering about the endstop location at the top of the elbow assembly. Looking at the CAD of the whole arm assembly, https://myhub.autodesk360.com/ue2a87a43/g/shares/SH919a0QTf3c32634dcf530f7dd13178bc27

They are both connected to same "rods" the one that moves the hotend. Is that correct? Should the top one be at the bottom instead? (You can see the 8mm steel rods, I've attached pictures to make it clearer as I suck at explaining things)
https://www.dropbox.com/s/c88cly4qcrnnkyl/2021-12-07 16.10.55.jpg?dl=0
https://www.dropbox.com/s/0eipmqwkznc4g37/2021-12-07 16.11.00.jpg?dl=0
https://www.dropbox.com/s/yo8shwanv4avas6/2021-12-07 16.11.10.jpg?dl=0
https://www.dropbox.com/s/7gzxbvur95vhifk/2021-12-07 16.11.25.jpg?dl=0
https://www.dropbox.com/s/k9c9zb6bw78sz6t/2021-12-07 16.11.25_LI.jpg?dl=0 (This one shows if the endstop should be moved to the one rod below).
I've build the whole thing, and this is the only thing that confuse me, because right now when I move the hotend arm, both endstops move, which I don't think should be the case? Thanks.

iamthebest22, 2021-12-09:
Nvm, I've asked other scara, can be either options,.

closed 2021-12-09 by iamthebest22

madl3x, 2021-12-16:
I know you closed this, but I will answer anyway, for posterity.
The shoulder (distal arm) driving axis should not move when you move the elbow (proximal arm).

That driving axis should be on top of the arm, where the shoulder starts. There is another axis colinear with this one, on the bottom of the arm, that is indeed driving the elbow, and is providing another 1/3 reduction. That is the one that you noticed moving when you move the tip. 

You should put the shoulder endstop where the shoulder driving axis is, the one that moves independently over the elbow.
Hope it helps.

iamthebest22, 2021-12-16:
thankyou, that really clears it up. I've flipped the orientation, than the CAD files showed, so yes, the top endstop is now on the shoulder endstop, and doesn't move when I move the hotend arm. the lower endstop is now on the one that moves the hotend arms. Thanks! Now to sort it out for RRF.