madl3x/x-scara #14: I wonder about direction of shoulder and elbow when press +X , -X and +Y, -Y on pronterface test

ToonKru3, 2022-02-11:
**Right Now I have use your X-scara firmware.I know how to set INVERT my motor but I confuse what is the correct direction.I test on Pronterface and Repetier-Host.My Question is**
[image]

1. When i press +X  or -X what direction could be ?
2. and +Y and -Y direction

ToonKru3, 2022-02-17:
and how to set up "start gcode" from repetier or other slicer program to start printer

madl3x, 2022-02-20:
[image]

madl3x, 2022-02-20:
`and how to set up "start gcode" from repetier or other slicer program to start printer

You could use the defaults for a regular cartesian printer. The only special code I use for a startup is lifting the nozzle up 15 mm before homing. That is because I have a heated bed that the arm could crash into when G28 is run.

Startup g-code:
```
;Home all axis
G28
;Move the arm away from print area
M361 S-90 E90
;Move the arm up 15mm
G1 Z15.0 F6000 
;Prime the extruder
G92 E0
G1 F200 E5
G92 E0
;Start printing..
```
End g-code:
```;Move the arm away from print area
G0 F3000 ; set feedrate
M361 S-90 E90
;Stop heaters
M104 S0
M140 S0
;Retract the filament
G92 E3
G1 E-3 F300
; Lower it 
G0 Z-6
; Disable steppers (arm will drop)
M84
```

ToonKru3, 2022-02-20:
this is for 0.4 beta firmware?  And i have another problem with wrong geometry curve trapezoid
[image]

ToonKru3, 2022-02-20:
And 0.4 firmwarm i have to calibrate m362 e0 s0 or not important on 0.4 just homing with endstop and print right?

ToonKru3, 2022-02-21:
Right now i stright my scara arm and then calibrate with gcode m362 e0 s0 and use g0 x0 y0 nest hotend is center of my robot but when I test with repetier host  X+ and X- button.they don't move stright line but curve line and when i test print squre it's curve again 

[image]

It's depending on "number teeth" of shoulder gear and elbow gear ? Like need to have even or odd number of teeth ?
 my scara I design with Shoulder 223 teeth and Elbow 144 teeth or not this case but something in firmware ?

ToonKru3, 2022-02-21:
This is my edit firmware

/**
 * Marlin 3D Printer Firmware
 * Copyright (c) 2020 MarlinFirmware [https://github.com/MarlinFirmware/Marlin]
 *
 * Based on Sprinter and grbl.
 * Copyright (c) 2011 Camiel Gubbels / Erik van der Zalm
 *
 * This program is free software: you can redistribute it and/or modify
 * it under the terms of the GNU General Public License as published by
 * the Free Software Foundation, either version 3 of the License, or
 * (at your option) any later version.
 *
 * This program is distributed in the hope that it will be useful,
 * but WITHOUT ANY WARRANTY; without even the implied warranty of
 * MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
 * GNU General Public License for more details.
 *
 * You should have received a copy of the GNU General Public License
 * along with this program.  If not, see <http://www.gnu.org/licenses/>.
 *
 */
#pragma once

/**
 * Configuration.h
 *
 * Basic settings such as:
 *
 * - Type of electronics
 * - Type of temperature sensor
 * - Printer geometry
 * - Endstop configuration
 * - LCD controller
 * - Extra features
 *
 * Advanced settings can be found in Configuration_adv.h
 *
 */
#define CONFIGURATION_H_VERSION 020005

//===========================================================================
//============================= Getting Started =============================
//===========================================================================

/**
 * Here are some standard links for getting your machine calibrated:
 *
 * http://reprap.org/wiki/Calibration
 * http://youtu.be/wAL9d7FgInk
 * http://calculator.josefprusa.cz
 * http://reprap.org/wiki/Triffid_Hunter%27s_Calibration_Guide
 * http://www.thingiverse.com/thing:5573
 * https://sites.google.com/site/repraplogphase/calibration-of-your-reprap
 * http://www.thingiverse.com/thing:298812
 */

//===========================================================================
//============================= SCARA Printer ===============================
//===========================================================================

/**
 * X-SCARA is developed by Alex Mircescu in 2021 and is an open source 
 * SCARA machine with an unique mechanical design, intented to simplify the
 * calibration process and to improve overall mechanical performance and 
 * robustness for a serial SCARA machine.
 * 
 * For more information:
 *    Thingiverse: https://www.thingiverse.com/thing:4500425
[... 2247 lines cut ...]

/**
 * R/C SERVO support
 * Sponsored by TrinityLabs, Reworked by codexmas
 */

/**
 * Number of servos
 *
 * For some servo-related options NUM_SERVOS will be set automatically.
 * Set this manually if there are extra servos needing manual control.
 * Leave undefined or set to 0 to entirely disable the servo subsystem.
 */
//#define NUM_SERVOS 3 // Servo index starts with 0 for M280 command

// (ms) Delay  before the next move will start, to give the servo time to reach its target angle.
// 300ms is a good value but you can try less delay.
// If the servo can't reach the requested position, increase it.
#define SERVO_DELAY { 300 }

// Only power servos during movement, otherwise leave off to prevent jitter
//#define DEACTIVATE_SERVOS_AFTER_MOVE

// Allow servo angle to be edited and saved to EEPROM
//#define EDITABLE_SERVO_ANGLES

ToonKru3, 2022-02-21:
This is my edit firmware

`/**
 * Marlin 3D Printer Firmware
 * Copyright (c) 2020 MarlinFirmware [https://github.com/MarlinFirmware/Marlin]
 *
 * Based on Sprinter and grbl.
 * Copyright (c) 2011 Camiel Gubbels / Erik van der Zalm
 *
 * This program is free software: you can redistribute it and/or modify
 * it under the terms of the GNU General Public License as published by
 * the Free Software Foundation, either version 3 of the License, or
 * (at your option) any later version.
 *
 * This program is distributed in the hope that it will be useful,
 * but WITHOUT ANY WARRANTY; without even the implied warranty of
 * MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
 * GNU General Public License for more details.
 *
 * You should have received a copy of the GNU General Public License
 * along with this program.  If not, see <http://www.gnu.org/licenses/>.
 *
 */
#pragma once

/**
 * Configuration.h
 *
 * Basic settings such as:
 *
 * - Type of electronics
 * - Type of temperature sensor
 * - Printer geometry
 * - Endstop configuration
 * - LCD controller
 * - Extra features
 *
 * Advanced settings can be found in Configuration_adv.h
 *
 */
#define CONFIGURATION_H_VERSION 020005

//===========================================================================
//============================= Getting Started =============================
//===========================================================================

/**
 * Here are some standard links for getting your machine calibrated:
 *
 * http://reprap.org/wiki/Calibration
 * http://youtu.be/wAL9d7FgInk
 * http://calculator.josefprusa.cz
 * http://reprap.org/wiki/Triffid_Hunter%27s_Calibration_Guide
 * http://www.thingiverse.com/thing:5573
 * https://sites.google.com/site/repraplogphase/calibration-of-your-reprap
 * http://www.thingiverse.com/thing:298812
 */

//===========================================================================
//============================= SCARA Printer ===============================
//===========================================================================

/**
 * X-SCARA is developed by Alex Mircescu in 2021 and is an open source 
 * SCARA machine with an unique mechanical design, intented to simplify the
 * calibration process and to improve overall mechanical performance and 
 * robustness for a serial SCARA machine.
 * 
 * For more information:
 *    Thingiverse: https://www.thingiverse.com/thing:4500425
[... 2248 lines cut ...]
/**
 * R/C SERVO support
 * Sponsored by TrinityLabs, Reworked by codexmas
 */

/**
 * Number of servos
 *
 * For some servo-related options NUM_SERVOS will be set automatically.
 * Set this manually if there are extra servos needing manual control.
 * Leave undefined or set to 0 to entirely disable the servo subsystem.
 */
//#define NUM_SERVOS 3 // Servo index starts with 0 for M280 command

// (ms) Delay  before the next move will start, to give the servo time to reach its target angle.
// 300ms is a good value but you can try less delay.
// If the servo can't reach the requested position, increase it.
#define SERVO_DELAY { 300 }

// Only power servos during movement, otherwise leave off to prevent jitter
//#define DEACTIVATE_SERVOS_AFTER_MOVE

// Allow servo angle to be edited and saved to EEPROM
//#define EDITABLE_SERVO_ANGLES
`

ToonKru3, 2022-02-21:
[image]
[image]
[image]