madl3x/x-scara #16: Marlin 2.0.9.3

Riverolf, 2022-04-16:
I had to download the newest marlin 2.0.9.3 to get my skr e3 v3 board to come up in the beard definitions.
When using this version, i follow the config readme file, and the line #define SCARA throws an error, saying that it is nor MORGAN_SCARA, So I change it to this. Is this correct?

I also cannot use the #define X_MAX_LENGTH 300 and #define Y_MAX_LENGTH 300 line. It throws an error.

Everytring else is able to compile,but I have not been able to test the machine yet. I am missing some parts.
Will this be okay to use, or do I need to change something more?

Riverolf, 2022-05-02:
I managed to get the printer up and running using MP_SCARA in marlin. But I cannot find how to define elbow crosstalk. I see that this has been taken care of in the X-SCARA code. Does anybody know how to do this in MP_SCARA?

gnmrc-s, 2022-05-03:
You have to add the crosstalk in the kinematics file, like it has been done in the x scara, follow the calculation that madl3x has done. 
I also tried with MP_SCARA configuration, by how to mechatronics but I wasn't as lucky as you, my arm is not stopped by the endstops during homing, even if they work properly.

Riverolf, 2022-05-03:
I am one step further with the MP_SCARA side of things. 
I changed the scara.cpp file in line 129:
  
// Angle of Arm2
    PSI = ATAN2(S2, C2);
    PSI = PSI + THETA/3; //RSTROMO 

I added the line where RSTROMO is commented. This seems to make the prints look more like they should. 
i will be testing it the next few days.